scross01/prometheus-klipper-exporter #2: 500 error when running "printer_objects" & "temperature" modules together

wywywywy, 2022-10-04:
When running the "printer_objects" & "temperature" modules together together, it returns the following error:

```
curl "localhost:9101/probe?modules=printer_objects&modules=temperature&target=192.168.1.39%3A7125"

An error has occurred while serving metrics:

6 error(s) occurred:
* collected metric klipper_extruder_temperature gauge:<value:21.15 >  has help "Klipper extruder temperature." but should have "Klipper extruder temperature"
* collected metric klipper_extruder_target gauge:<value:0 >  has help "Klipper extruder target." but should have "Klipper extruder target"
* collected metric klipper_extruder_power gauge:<value:0 >  has help "Klipper extruder power." but should have "Klipper extruder power"
* collected metric klipper_heater_bed_temperature gauge:<value:21.27 >  has help "Klipper heater bed temperature." but should have "Klipper heater_bed temperature"
* collected metric klipper_heater_bed_target gauge:<value:0 >  has help "Klipper heater bed target." but should have "Klipper heater_bed target"
* collected metric klipper_heater_bed_power gauge:<value:0 >  has help "Klipper heater bed power." but should have "Klipper heater_bed power"
```

However, when they are run individually, they work fine.

```
curl "localhost:9101/probe?modules=printer_objects&target=192.168.1.39%3A7125"
curl "localhost:9101/probe?modules=temperature&target=192.168.1.39%3A7125"
```

scross01, 2022-10-04:
Confirmed, duplicate metric names. Actually the same metrics just from a different API endpoint.  I'm thinking the temperature module can be removed as `printer_objects` is mostly a superset.  Only difference is the `temperature` module will get additional `temperature_sensor` data - this still needs to be added to `printer_objects`. 

Created Issue #3 to track update to `printer_objects`. Added note to `README.md`. Will leave this open for now and close when `temperature` module is removed.

scross01, 2022-12-28:
`temperature` module deprecated and removed from documentation as of `v0.8.0`

closed 2022-12-28 by scross01