scross01/prometheus-klipper-exporter #3: add `temperature_sensor` metrics to `printer_objects` module

scross01, 2022-10-04:
Add metric collection for custom `temperature_sensor` objects. 

https://moonraker.readthedocs.io/en/latest/printer_objects/

## temperature_sensor sensor_name
Enabled when `[temperature_sensor sensor_name]` is included in `printer.cfg`. It is possible for multiple temperature sensors to be configured.
```
{
    "temperature": 0.0,
    "measured_min_temp": 0.0,
    "measured_max_temp": 0.0
}
`
A temperature_sensor reports the following state:
`temperature`: Sensor's current reported temperature
`measured_min_temp`: The mimimum temperature read from the sensor
`measured_max_temp`: The maximum temperature read from the sensor

Ramalama2, 2022-10-15:
@scross01 Sry to nerv again,
may i ask how exactly i can add my own cutom temperature sensor? I mean sth like an example?
I have 4 Frame sensors / Chamber sensor etc...

```
scrape_configs:

  - job_name: "voron2"
    scrape_interval: 5s
    metrics_path: /probe
    static_configs:
      - targets: [ 'voron2.user.mwpp.eu:7125' ]
    params:
      modules: [ "process_stats", "job_queue", "system_info", "network_stats", "directory_info", "printer_objects" ]
    relabel_configs:
      - source_labels: [__address__]
        target_label: __param_target
      - source_labels: [__param_target]
        target_label: instance
      - target_label: __address__
        replacement: docker.mwpp.eu:9101

  # optional exporter metrics
  - job_name: "klipper-exporter"
    scrape_interval: 5s
    metrics_path: /metrics
    static_configs:
      - targets: [ 'docker.mwpp.eu:9101' ]
```

Thats the example config :-)

Thank you!

scross01, 2022-10-16:
Release `v0.6.0` adds support for `temperature_sensor`, `temperate_fan`, and `output_pin` metrics

commit 981901d7 (2022-10-16): Fixes #3, Fixes #4, Fixes #5. Add support for custom sensors.
closed 2022-10-16 by scross01