scross01/prometheus-klipper-exporter #6: Feature request: Add Docker image

wywywywy, 2022-10-04:
It'd be really useful to have it as a docker image so we can easily put it into k8s etc.

I've got a ready-made `Dockerfile` for you, but I don't have time to create a PR right now.

```
## Build
FROM golang:1.19 AS build

WORKDIR /app

COPY . ./

RUN make build

## Deploy
FROM gcr.io/distroless/base-debian11

WORKDIR /

COPY --from=build /app/prometheus-klipper-exporter /prometheus-klipper-exporter

EXPOSE 9101

USER nonroot:nonroot

ENTRYPOINT ["/prometheus-klipper-exporter"]
```

scross01, 2022-10-04:
Thanks! Have something very similar put together already :-)
Next step is looking at creating/publishing a multi-platform image.

wywywywy, 2022-10-04:
Nice one!

Here's an example GitHub workflow to create & publish multi-platform Docker images: https://github.com/wywywywy/tesla-prometheus-exporter/blob/master/.github/workflows/docker-publish.yml

scross01, 2022-10-05:
Thanks again @wywywywy, that really helped. `Dockerfile` added, and workflow action used to publish to GHCR.

https://github.com/scross01/prometheus-klipper-exporter/pkgs/container/prometheus-klipper-exporter

closed 2022-10-05 by scross01