scross01/prometheus-klipper-exporter #13: more metrics

badsmoke, 2022-12-19:
Hey,

thanks for this great project, i recently switched to klipper and am a big metrics fan.

metrics that would still interest me would be 

current flow
current speed

there are at least 3 values for the current print time but none for how long the print should take in total according to the print time

heightmap, since I create a bed mesh before each print, one could collect these values and compare them in the long term.

perhaps also the mcu utilization would be interesting.

thanks

scross01, 2022-12-20:
Thanks for the suggestions. Some of these would require changes to Klipper/Moonraker to expose the metric data through the APIs. 

Some thoughts on each:

**current flow/speed** - doesn't seem to be possible with the current API. Sampling the values every poll may not be that useful as the flow and speed can change so quickly. The min, max an average in the polling period would probably be more useful than the discreet values.

**failed/aborted/successful prints** -  will look at adding this, should be possible by querying job history and processing the results. Not the most efficient approach, would be better if Moonraker exposed these as metrics.

**estimated print time** - not exposed by the main Moonraker APIs, but is in the Octoprint compatible API output, so should be possible to add.

**heightmap** - need to think about this one as Prometheus isn't really geared to collecting datapoint for a 2d mesh. Maybe a label for each point e.g. `klipper_heighmap_offset{point="x1y1"}`

**mcu utilization** - the only metric I get for the mcu on my printer is the temperature (Ender3 v2 with v4.2.2 board), maybe it varies by board?  For metrics about the host that is running Klipper I use the https://github.com/prometheus/node_exporter

scross01, 2022-12-22:
No luck getting the estimated print time from the OctoPrint compatible API - The [job status API](https://github.com/Arksine/moonraker/blob/dde9bcc752c4827ff0ffc01ed99c6f487122c04f/moonraker/components/octoprint_compat.py#L258) is not fully implemented and returns an empty payload for all values except the printer state 😞

scross01, 2023-01-13:
v0.9.0 adds several MCU specific metrics to the `printer_objects` module

`klipper_mcu_awake`
`klipper_mcu_clock_frequency`
`klipper_mcu_invalid_bytes`
`klipper_mcu_read_bytes`
`klipper_mcu_ready_bytes`
`klipper_mcu_receive_seq`
`klipper_mcu_retransmit_bytes`
`klipper_mcu_retransmit_seq`
`klipper_mcu_rto`
`klipper_mcu_rttvar`
`klipper_mcu_send_seq`
`klipper_mcu_stalled_bytes`
`klipper_mcu_srtt`
`klipper_mcu_write_bytes`