scross01/prometheus-klipper-exporter #15: Add Authentication

jangrewe, 2022-12-27:
Thank you for creating this exporter! I just moved away from an Octoprint-based image and had metrics available via a plugin there, but now on FluiddPi there was nothing comparable...

The issue i'm having is that i need to enforce authentication, but i can't seem to figure out how to use an API Key with Prometheus and your exporter... could you add something to allow passing an API key?

scross01, 2022-12-28:
Release `v0.8.0` adds options to set the API Key for authentication in either the `prometheus.yml` config, using a `-moonraker.apikey` command line option, or setting a `MOONRAKER_APIKEY` environment variable.

See the [Authentication](https://github.com/scross01/prometheus-klipper-exporter#authentication) section in the README for usage details.

closed 2022-12-28 by scross01

jangrewe, 2022-12-28:
Wow, that was fast - Thank you! 😍 

You don't happen to feel like also sharing your Grafana dashboard source, do you? 😁

scross01, 2022-12-28:
I've added an example dashboard that can be imported as a starting point. 
https://github.com/scross01/prometheus-klipper-exporter/tree/main/example#example-dashboard

jangrewe, 2022-12-28:
Beautiful, thank you! 😄