scross01/prometheus-klipper-exporter #16: Not familiar with prometheus

Leviathan3DPrinting, 2022-12-29:
I am trying to get this running in docker. I'm vaguely familiar with all of this but your project caught my attention! I would love to log a lot of statistics but am confused as to how to actually set this up. It's probably something simple I am not doing correctly.

My error:
Get "http://10.0.0.222:9101/probe?modules=process_stats&modules=network_stats&modules=system_info&modules=job_queue&modules=directory_info&modules=printer_objects&modules=history&target=10.0.0.222%3A7125": dial tcp 10.0.0.222:9101: connect: connection refused

scross01, 2022-12-30:
From error it looks like Prometheus failing to connect to the klipper-exporter process at 10.0.0.222:9101.  Your target for Klipper is 10.0.0.222:7125, so I'm assuming your running the exporter and Klipper on the same host. Where is your Docker environment and Prometheus running? is it also on the same host, or a different environment. 

```mermaid
graph LR
  Prometheus --> Exporter[Klipper-Exporter<br/>10.0.0.222:9101]
  Exporter --> Klipper[Klipper<br/>10.0.0.222:7125]
```

Leviathan3DPrinting, 2022-12-30:
Ok so after some more digging I got it working by using the docker container id as the ip address as such ac6a583c6268:9101

I had some questions regarding prometheus. I'm vaguely familiar with influxDB. Is the docker image setup to store data recursively? Is there a way to bake in averaging of the data like say I want to know the average bed power accross an entire print? Then say take the average across each print in a month? If you have any resources you recommend I read or watch I will happily do my homework I am just confused as to where to start. I did however get this running!

scross01, 2022-12-31:
Take a look at the [PromQL query syntax](https://prometheus.io/docs/prometheus/latest/querying/basics/), its rather different from Influx queries.

The Explore option in Grafana can be useful for testing queries, using the query builder to help build queries with functions for averages, rate of change, etc.

Leviathan3DPrinting, 2023-01-08:
I ran into issues running this on my pi zero 2 w. I'm not sure if it's the pi or the SD card. I have a pi 4 I can try but I was wondering if you ran into any similar issues with klipper crashing. I am running docker but I don't think that's the issue. I setup some pretty cool logging before I started getting crashes. I think for now I will just set up prometheus with a specified amount of storage it seems to take up a lot less data than I was expecting.

Also thanks for the help. You have saved me so much headache with this repo!

Hopefully you enjoy this marco!

[gcode_macro SEARCH_VARS]
gcode:
    {% set search = params.S|lower %}
    {% set ns = namespace() %}
    {% for item in printer  %}
        {% if ' ' in item %}
            {% set ns.path = ['printer', "['%s']" % (item), ''] %}
        {% else %}
            {% set ns.path = ['printer.', item, ''] %}   
        {% endif %} 

        {% if search in ns.path|lower %}
            { action_respond_info(ns.path|join) }
        {% endif %} 

        {% if printer[item].items() %}
            {% for childkey, child in printer[item].items() recursive %}
                {% set ns.path = ns.path[:loop.depth|int + 1] %}

                {% if ' ' in childkey %}
                    {% set null = ns.path.append("['%s']" % (childkey)) %}
                {% else %}
                    {% set null = ns.path.append(".%s" % (childkey)) %}
                {% endif %} 

                {% if child is mapping  %}
                    { loop(child.items()) }
                {% else %}
                    {% if search in ns.path|lower %}
                        { action_respond_info("%s : %s" % (ns.path|join, child)) }   
                    {% endif %} 
                {% endif %} 
                
            {% endfor %}
        {% endif %} 
    {% endfor %}

Leviathan3DPrinting, 2023-01-08:
Be careful it spits out a lot!

scross01, 2023-01-08:
I have Klipper and the exporter running on a Pi Zero 2 W as well, but I run Prometheus and Grafana on a separate Pi 4.  The Pi Zero doesn't really have enough memory.  I also changed my Docker/Prometheus configuration on the Pi 4 so the data is stored on external storage rather than the SD card, as the constant writes could eventually kill the card.

scross01, 2023-01-08:
Cool macro, I made one small tweak to prevent it outputting the full gcode of macros it finds.
Change `{% if search in ns.path|lower %}` to `{% if search in ns.path|lower and '.gcode' not in ns.path %}`

scross01, 2025-08-19:
closing old issue

closed 2025-08-19 by scross01