scross01/prometheus-klipper-exporter #24: Panic when the network in unreachable

ctrochalakis, 2024-05-11:
Hello and thank you for klipper exporter.

I am getting the following error when the network is unreachable. I am using tailscale as a VPN which can be unavailable.

```
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]: time="2024-05-11T07:50:30+03:00" level=error msg="Get \"http://my-printer:7125/server/history/totals\": dial tcp 100.98.26.56:7125: connect: network is unreachable"
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]: panic: runtime error: invalid memory address or nil pointer dereference
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x8068e0]
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]: goroutine 41 [running]:
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]: github.com/scross01/prometheus-klipper-exporter/collector.collector.Collect({{0x9ae530, 0xc0000b4370}, {0xc0000a2078, 0x17}, {0xc0000ac180, 0x2, 0x2}, {0x0, 0x0}, {{0x9aa4e0, ...}, ...}}, ...)
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]:         github.com/scross01/prometheus-klipper-exporter/collector/collector.go:201 +0x18e0
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]: github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]:         github.com/prometheus/client_golang@v1.13.0/prometheus/registry.go:453 +0x102
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]: created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather in goroutine 38
May 11 07:50:30 yix prometheus-klipper-exporter[3777169]:         github.com/prometheus/client_golang@v1.13.0/prometheus/registry.go:464 +0x568
May 11 07:50:30 yix systemd[1]: klipper_exporter.service: Main process exited, code=exited, status=2/INVALIDARGUMEND
```

scross01, 2024-05-11:
Reproduced (similar) issue by disabling and enabling the local network connection

```
klipper-exporter  | time="2024-05-11T11:43:25Z" level=error msg="unexpected EOF"
klipper-exporter  | panic: runtime error: invalid memory address or nil pointer dereference
klipper-exporter  | [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x3e3af4]
klipper-exporter  | 
klipper-exporter  | goroutine 1724 [running]:
klipper-exporter  | github.com/scross01/prometheus-klipper-exporter/collector.Collector.Collect({{0x595568, 0x40003512c0}, {0x400000ee74, 0xc}, {0x40002f2800, 0x8, 0x8}, {0x0, 0x0}}, 0x40002e09c0)
klipper-exporter  |     /app/collector/collector.go:167 +0x1074
klipper-exporter  | github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
klipper-exporter  |     /go/pkg/mod/github.com/prometheus/client_golang@v1.13.0/prometheus/registry.go:453 +0xd8
klipper-exporter  | created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather in goroutine 1448
klipper-exporter  |     /go/pkg/mod/github.com/prometheus/client_golang@v1.13.0/prometheus/registry.go:464 +0x4a8
klipper-exporter exited with code 0
```

The underlying cause appears to be in the prometheus/client_golang library. Bumping from v1.13.0 to latest version v0.19.1

scross01, 2024-05-11:
Fixed in release v0.10.2

closed 2024-05-11 by scross01

ctrochalakis, 2024-07-10:
Hello again, I am having the same issue when the node is off, triggering an i/o timeout and eventually a nil pointer dereference.

```
Jul 09 23:55:05 yix prometheus-klipper-exporter[194451]: time="2024-07-09T23:55:05+03:00" level=error msg="Get \"http://my-printer:7125/server/history/totals\": dial tcp [fd7a:115c:a1e0:ab12:4843:cd96:6262:1a38]:7125: i/o timeout"
Jul 09 23:55:05 yix prometheus-klipper-exporter[194451]: panic: runtime error: invalid memory address or nil pointer dereference

scross01, 2024-07-11:
Should be fixed with v0.11.1