scross01/prometheus-klipper-exporter #26: json: cannot unmarshal number 1320.0 into Go struct field .result.job_totals.total_jobs of type int64

jangrewe, 2024-06-02:
Hi, i've been running your exporter for quite a while (thanks for it, by the way!), but since a couple of days i'm running into this issue:

```
time="2024-06-02T11:53:40Z" level=info msg="Beginning to serve on port :9101"
time="2024-06-02T11:53:43Z" level=info msg="Starting metrics collection of [process_stats network_stats job_queue system_info directory_info printer_objects history] for voron0box.faked.lan:7125"
time="2024-06-02T11:53:43Z" level=info msg="Collecting process_stats for voron0box.faked.lan:7125"
time="2024-06-02T11:53:43Z" level=info msg="Collecting directory_info for voron0box.faked.lan:7125"
time="2024-06-02T11:53:43Z" level=info msg="Collecting job_queue for voron0box.faked.lan:7125"
time="2024-06-02T11:53:43Z" level=info msg="Collecting history for voron0box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=info msg="Collecting active print for voron0box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=info msg="Collecting system_info for voron0box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=info msg="Collecting printer_objects for voron0box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=info msg="Found custom sensors: &[RaspberryPi Controller Picobilical chamber] &[] &[beeper]"
time="2024-06-02T11:53:44Z" level=info msg="Starting metrics collection of [process_stats network_stats job_queue system_info directory_info printer_objects history] for voron2box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=info msg="Collecting process_stats for voron2box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=info msg="Collecting directory_info for voron2box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=info msg="Collecting job_queue for voron2box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=info msg="Collecting history for voron2box.faked.lan:7125"
time="2024-06-02T11:53:44Z" level=error msg="json: cannot unmarshal number 1320.0 into Go struct field .result.job_totals.total_jobs of type int64"
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x79032d]
goroutine 34 [running]:
github.com/scross01/prometheus-klipper-exporter/collector.Collector.Collect({{0x912240, 0xc00013ad20}, {0xc0001547e0, 0x18}, {0xc00012a580, 0x7, 0x8}, {0xc000130720, 0x20}}, 0xc00034bce0)
        /app/collector/collector.go:195 +0x1aad
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
        /go/pkg/mod/github.com/prometheus/client_golang@v1.19.1/prometheus/registry.go:455 +0x105
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather in goroutine 31
        /go/pkg/mod/github.com/prometheus/client_golang@v1.19.1/prometheus/registry.go:466 +0x568
        
```

As you can see, one of the printers gets scraped successfully, but then second one runs into an error.

My Prometheus scrape target config follows your example, and i'm not setting any arguments besides the log level (typically Warning, but currently Info to find out what's wrong).

jangrewe, 2024-06-02:
Oh, and one little feature request: Could you change the Dockerfile from
```
CMD ["./main"]
```
to
```
ENTRYPOINT ["./main"]
```

This would allow setting any arguments like `-logging.level=Warning` without having to set it to `/root/main -logging.level=Warning`, as providing arguments to the container will override `CMD`, but it will keep the `ENTRYPOINT`.

jangrewe, 2024-06-08:
@scross01 I think only this line needs to be changed from `int64` to `float64`: https://github.com/scross01/prometheus-klipper-exporter/blob/e27d9e2f0cd60f2618f6d91706fa14c8ffdff7d8/collector/history.go#L15

scross01, 2024-06-08:
Thanks for finding the fix for this. I just updated my Moonraker version and see the same issue. Looks like the recent change in Moonraker caused the API to start returning a float instead of an in for the `total_jobs`, probably caused by the switch from using lldb to sqlite https://github.com/Arksine/moonraker/pull/858

scross01, 2024-06-08:
Fix in v0.10.3

closed 2024-06-08 by scross01

jangrewe, 2024-06-08:
Thanks for responding so quickly, the exporter is now running fine again!

Another issue, besides the `CMD` vs. `ENTRYPOINT` mentioned above: I run the exporter with
```
ghcr.io/scross01/prometheus-klipper-exporter:latest  /root/main -logging.level=Warning
```
but i still get log messages for `Info` - and that amount adds up quickly.

scross01, 2024-06-08:
Try just

```shell
docker run ghcr.io/scross01/prometheus-klipper-exporter:latest -logging.level=Warning
```

or 

```shell
docker run --entrypoint "/root/main" ghcr.io/scross01/prometheus-klipper-exporter:latest -logging.level=Warning
```

jangrewe, 2024-06-08:
Sorry, i completely missed that you also changed `CMD` to `ENTRYPOINT` already, all works as expected now! 👌🏻