scross01/prometheus-klipper-exporter #30: API Key generation

Nikongen, 2024-07-09:
Hi,

described method for generating the api key did not work for me.
First I got a package not found error (lmdb)
After installing this to the moonraker env I get a file not found error.
Note I installed klipper and moonraker with kiauh on Ubuntu Server .
What worked for me was:

`curl localhost:7125/access/api_key`

scross01, 2024-07-09:
Thanks for reporting this.  The guidelines are based on the Moonraker docs [Retrieving the API Key](https://moonraker.readthedocs.io/en/latest/installation/#retrieving-the-api-key), but it looks like the script may have stopped working for newer installations, the `fetch-apikey.sh` script uses the lmdb package, but there's a note further down in the docs that mentions LMDB Database is deprecated.

Using the curl approach seems to be more reliable.

Nikongen, 2024-07-09:
Yes, took me a while to find the curl method... 

Btw, thanks for the nice project. I am using this inside a podman Kube/pod and it works like a charm. 🤩👌

If interested I can also share my yaml for the kube.

scross01, 2025-08-21:
Closing this issue. 

The documented approach using `fetch-apikey.sh` is working again.  Seems to have been fixed in Moonraker at some point in the last year.

closed 2025-08-21 by scross01