scross01/prometheus-klipper-exporter #32: Can't read history on fresh install

michmela44, 2024-08-03:
I've just wiped my host and have a fresh Klipper install that has not run any jobs yet

I setup a docker enviromnent using the example as a guide

My result from http://klipper:7125/server/history/totals

looks like

```
{
  "result": {
    "job_totals": {
      "total_jobs": 0,
      "total_time": 0,
      "total_print_time": 0,
      "total_filament_used": 0,
      "longest_job": 0,
      "longest_print": 0
    },
    "auxiliary_totals": []
  }
}
```

this results in the following error when trying to start the klipper-exporter container

```
2024-08-02 21:17:11 time="2024-08-03T02:17:11Z" level=info msg="Collecting history for klipper:7125"
2024-08-02 21:17:11 time="2024-08-03T02:17:11Z" level=error msg="json: cannot unmarshal number 0.0 into Go struct field .result.job_totals.total_jobs of type int64"
2024-08-02 21:17:11 panic: runtime error: invalid memory address or nil pointer dereference
2024-08-02 21:17:11 [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x8228dd]
2024-08-02 21:17:11 
2024-08-02 21:17:11 goroutine 36 [running]:
2024-08-02 21:17:11 github.com/snelstim/klipper-exporter/collector.Collector.Collect({{0x9d3390, 0xc0003c20a0}, {0xc0003d0030, 0x14}, {0xc0003a4080, 0x8, 0x8}, {0xc00002a011, 0x22}}, 0xc0003c6120)
2024-08-02 21:17:11     /app/collector/collector.go:203 +0x1a7d
2024-08-02 21:17:11 github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
2024-08-02 21:17:11     /go/pkg/mod/github.com/prometheus/client_golang@v1.13.0/prometheus/registry.go:453 +0x105
2024-08-02 21:17:11 created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather in goroutine 33
2024-08-02 21:17:11     /go/pkg/mod/github.com/prometheus/client_golang@v1.13.0/prometheus/registry.go:464 +0x568
```

then the klipper-exporter container crashes and restarts

michmela44, 2024-08-03:
Looks like this may have been fixed here

https://github.com/scross01/prometheus-klipper-exporter/commit/6d86e5311e67d0e2a72e14433ff2183cb2ad02aa

But the docker image is not updated

michmela44, 2024-08-03:
Ok, I'm not sure what happened, but somehow my docker.yml file had a different source for the image in it.

After updating the docker.yml file to 

```
 klipper-exporter:
    hostname: klipper-exporter
    image: ghcr.io/scross01/prometheus-klipper-exporter:latest

```

the history module seems to be working

closed 2024-08-03 by michmela44

scross01, 2024-08-04:
Glad to hear you got it working