scross01/prometheus-klipper-exporter #33: Issue using hostname

michmela44, 2024-08-05:
I had 2 printers set up using IP address.

I decided I wanted it to use host name to make it easier to pick which printer I wanted to view data for.

As a test, I update one of my printers to use the host name instead of IP following the example 

```
scrape_configs:
  - job_name: "klipper"
    scrape_interval: 5s
    metrics_path: /probe
    static_configs:
      - targets: [ '192.168.100.104:7125', 'ender3pro.local:7125' ]
    params:
      modules: [
        'process_stats',
        'network_stats',
        'system_info',
        'job_queue',
        'directory_info',
        'printer_objects',
        'history'
      ]
    relabel_configs:
      - source_labels: [__address__]
        target_label: __param_target
      - source_labels: [__param_target]
        target_label: instance
      - target_label: __address__
        replacement: klipper-exporter:9101
```

Prometheus shows that both are "UP" but the data doesn't seem to be coming in.

[image]

I checked the klipper-exporter logs and see

```
2024-08-04 21:40:22 time="2024-08-05T02:40:22Z" level=info msg="Collecting process_stats for ender3pro.local:7125"
2024-08-04 21:40:22 time="2024-08-05T02:40:22Z" level=error msg="unable to complete HTTP client request. Get \"http://ender3pro.local:7125/machine/proc_stats\": dial tcp: lookup ender3pro.local on 127.0.0.11:53: no such host"
```
I have my Klipper-exporter container set up like the example, and has my local DNS server set
```
  klipper-exporter:
    hostname: klipper-exporter
    image: ghcr.io/scross01/prometheus-klipper-exporter:latest
    container_name: klipper-exporter
    restart: unless-stopped
    expose:
      - 9101
    ports:
      - 9101:9101
    dns:
       192.168.100.1 # Or whatever your DNS is
```

http://Ender3Pro.local does work from the Docker host machine, and from other devices on the network, just not from inside the container.

Any ideas why this might not work?

scross01, 2024-08-05:
I think the issue here is because the `.local` mDNS addresses are not resolved within the docker network because multicast isn't bridged across the networks. Try removing the .local from the hostname and just use `ender3pro:7125` which should resolve through regular DNS.

michmela44, 2024-08-05:
Thanks, I'll look into more later today.

It looked like in the example you have, the ```.local``` existed on yours and worked, so I assumed it would do the same.

I would also be ok if there is a way to just change the labels in a hardcoded way in the yml file if that's possible? I tried something like this that I found online:

```
      static_configs:
      - targets: [ '192.168.100.104:7125' ]
        labels:
          instance: 'Ender 3 Pro'
```

But that didn't work either

scross01, 2024-08-05:
Ah yes, the docs are wrong, my bad.  I actually have my local network setup using a `.home.lan` domain for the local DNS, I had assumed `.local` would also work, when I wrote that. 

Will take a look label substitution as well.

scross01, 2024-08-05:
I found a way to add printer name as an additional label based on this example https://stackoverflow.com/a/65955484/1876039

Modify the `targets` to include the host IP address and the printer name, separated by an `@`
```yml
    static_configs:
      - targets: [ '192.168.50.248:7125@Ender-3-V2', '192.168.50.249:7125@Ender-3-Pro']
``` 
In the `relabel_configs` section replace original the `source_labels: [__address__]` with two entries below which extracts the IP address to the regular `instance` label, and adds a separate `printer` label with the printer name
```yml
    relabel_configs:
      - source_labels: [__address__]
        regex: '(.*)@.*'
        replacement: $1
        target_label: __param_target
      - source_labels: [__address__]
        regex: '.*@(.*)'
        replacement: $1
        target_label: printer
        ...
```

michmela44, 2024-08-05:
I have tried to do what I think you said to do, but it doesn't seem to work.

Is this how you meant for it to be?

```
scrape_configs:
  - job_name: "klipper"
    scrape_interval: 5s
    metrics_path: /probe
    static_configs:
      - targets: [ '192.168.100.104:7125', '192.168.100.110:7125@Ender-3-Pro' ]
    params:
      modules: [
        'process_stats',
        'network_stats',
        'system_info',
        'job_queue',
        'directory_info',
        'printer_objects',
        'history'
      ]
    relabel_configs:
      - source_labels: [__address__]
        regex: '(.*)@.*'
        replacement: $1
        target_label: __param_target
      - source_labels: [__address__]
        regex: '.*@(.*)'
        replacement: $1
        target_label: printer
      - source_labels: [__param_target]
        target_label: instance
      - target_label: __address__
        replacement: klipper-exporter:9101
```

scross01, 2024-08-06:
That looks right for the relabeling, but I think you'll also need to add a `@name` to first printer in the target list as well.

michmela44, 2024-08-06:
I did when I first tested it. I just forgot to put it in the example. 

When I ran it the first time it looked like the printer alias got appended to the job name somehow. 

I will try it again when my current print is finished.

michmela44, 2024-08-06:
Ok, 

I got back to testing it, and I found the issue. What I wanted was the text string applied to the "instance" label, but it was actually being applied to a label called "printer"

So I changed the config like this:
[image]

and now it is working correctly
[image]
[image]

Thank you very much for the help getting this working.

closed 2024-08-15 by scross01