scross01/prometheus-klipper-exporter #34: BTT PI not working

morkster, 2024-08-15:
My BTT PI is not sending printer_objects. I have confirmed regular PIs are working. Error message:
ERRO[0024] unable to unmarshal response data to collector.PrinterObjectResponse. unexpected end of JSON input

scross01, 2024-08-15:
Can you try running the API request directly on the command line to see what the JSON response is for this device.  Replace `klipperHost` with the hostname or IP of your BTT PI.

```shell
curl http://klipperHost/printer/objects/list
```

morkster, 2024-08-15:
{"error": {"code": 401, "message": "Unauthorized", "traceback": "Traceback
(most recent call last):\n\n  File
\"/home/mark/moonraker-env/lib/python3.9/site-packages/tornado/web.py\",
line 1769, in _execute\n    result = await result  # type: ignore\n\n  File
\"/home/mark/moonraker/moonraker/components/application.py\", line 503, in
prepare\n    self.current_user = await auth.authenticate_request(\n\n  File
\"/home/mark/moonraker/moonraker/components/authorization.py\", line 896,
in authenticate_request\n    raise HTTPError(401,
\"Unauthorized\")\n\ntornado.web.HTTPError: HTTP 401: Unauthorized
***@***.***:~#

{"result": {"objects": ["gcode", "webhooks", "configfile", "mcu", "mcu
rpi", "mcu displayEncoder", "menu", "display_status", "nopixel
displayStatus", "stepper_enable", "tmc2209 stepper_x", "tmc2209 stepper_y",
"tmc2209 stepper_z", "tmc2209 extruder", "heters", "heater_bed",
"temperature_sensor Chamber_Temp", "fan", "heater_fan hotend_fan",
"heater_fan board_fan", "neopixel my_nepixel", "idle_timeout",
"bed_screws", "gcode_move", "print_stats", "virtual_sdcard",
"pause_resume", "bed_mesh", "probe", "scres_tilt_adjust", "gcode_macro
_User_Variables", "gcode_macro _Probe_Variables", "gcode_macro
_klicky_check_variables_version", "code_macro _exit_point", "gcode_macro
_entry_point", "gcode_macro _Homing_Variables", "gcode_macro
Attach_Probe_Lock", "gcode_mcro Dock_Probe_Unlock", "gcode_macro
_Probe_Unlock", "gcode_macro _Probe_Lock", "gcode_macro Attach_Probe",
"gcode_macro Dock_Pobe", "gcode_macro PROBE_CALIBRATE", "gcode_macro
PROBE_ACCURACY", "gcode_macro _Umbilical_Path", "gcode_macro _Home_Z",
"gcodemacro _CheckProbe", "gcode_macro _SetProbeState", "gcode_macro
_Park_Toolhead", "gcode_macro BED_MESH_CALIBRATE", "gcode_macro
CREWS_TILT_CALCULATE", "gcode_macro PRINT_START", "gcode_macro PRINT_END",
"gcode_macro PAUSE", "gcode_macro RESUME", "gcode_maro CANCEL_PRINT",
"gcode_macro M600", "gcode_macro M105", "gcode_macro UNLOAD_FILAMENT",
"gcode_macro LEDOFF", "gcode_macro LEDIN", "gcode_macro LEDMAX",
"gcode_macro TILT_CALCULATE", "motion_report", "query_endstops",
"system_stats", "manual_probe", ***@***.***:~#

-- 
Met vriendelijke groet,

Best regards,

Mark Kroes

Cell     +31 (0) 6 244 44 883
E-mail ***@***.***

All information contained in this e-mail message may be confidential and is
intended to be exclusively for the addressee. Should you receive this
e-mail message unintentionally, please do not use the contents herein and
notify the sender immediately by return e-mail. Thank you.

morkster, 2024-08-15:
Also, if I run the docker, all of a sudden the container crashes with below errors:
time="2024-08-15T20:26:43Z" level=info msg="Collecting process_stats for 192.168.1.62:7125"
time="2024-08-15T20:26:43Z" level=info msg="Collecting directory_info for 192.168.1.62:7125"
time="2024-08-15T20:26:43Z" level=info msg="Collecting job_queue for 192.168.1.62:7125"
time="2024-08-15T20:26:43Z" level=info msg="Collecting history for 192.168.1.62:7125"
time="2024-08-15T20:26:43Z" level=info msg="Collecting active print for 192.168.1.62:7125"
time="2024-08-15T20:26:43Z" level=info msg="Collecting system_info for 192.168.1.62:7125"
time="2024-08-15T20:26:43Z" level=info msg="Collecting printer_objects for 192.168.1.62:7125"
time="2024-08-15T20:26:43Z" level=info msg="Found custom sensors: &[Chamber_Temp] &[] &[] &[] &[] &[]"
time="2024-08-15T20:26:44Z" level=info msg="Starting metrics collection of [process_stats job_queue system_info network_stats directory_info printer_objects history] for 192.168.7.42:7125"
time="2024-08-15T20:26:44Z" level=info msg="Collecting process_stats for 192.168.7.42:7125"
panic: runtime error: index out of range [-1]

goroutine 51 [running]:
github.com/scross01/prometheus-klipper-exporter/collector.Collector.Collect({{0x915220, 0xc0002aa230}, {0xc000286828, 0x11}, {0xc00028e100, 0x7, 0x8}, {0x0, 0x0}}, 0xc000382420)
        /app/collector/collector.go:61 +0x612e
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
        /go/pkg/mod/github.com/prometheus/client_golang@v1.19.1/prometheus/registry.go:455 +0x105
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather in goroutine 9
        /go/pkg/mod/github.com/prometheus/client_golang@v1.19.1/prometheus/registry.go:466 +0x568

morkster, 2024-08-15:
The actual service on the machine provides similar error:

INFO[0001] Starting metrics collection of [process_stats job_queue system_info network_stats directory_info printer_objects history] for 192.168.1.62:7125
INFO[0001] Collecting process_stats for 192.168.1.62:7125
INFO[0001] Collecting directory_info for 192.168.1.62:7125
INFO[0001] Collecting job_queue for 192.168.1.62:7125
INFO[0001] Collecting history for 192.168.1.62:7125
INFO[0001] Collecting active print for 192.168.1.62:7125
INFO[0001] Collecting system_info for 192.168.1.62:7125
INFO[0001] Collecting printer_objects for 192.168.1.62:7125
INFO[0001] Found custom sensors: &[Chamber_Temp] &[] &[] &[] &[] &[]
INFO[0002] Starting metrics collection of [process_stats job_queue system_info network_stats directory_info printer_objects history] for 192.168.7.42:7125
INFO[0002] Collecting process_stats for 192.168.7.42:7125
panic: runtime error: index out of range [-1]

goroutine 15 [running]:
github.com/scross01/prometheus-klipper-exporter/collector.Collector.Collect({{0x91c0a0, 0xc00007ce10}, {0xc00002a660, 0x11}, {0xc000155180, 0x7, 0x8}, {0x0, 0x0}}, 0xc00007b980)
        /root/klipper-exporter/prometheus-klipper-exporter/collector/collector.go:61 +0x612e
github.com/prometheus/client_golang/prometheus.(*Registry).Gather.func1()
        /root/go/pkg/mod/github.com/prometheus/client_golang@v1.19.1/prometheus/registry.go:455 +0x105
created by github.com/prometheus/client_golang/prometheus.(*Registry).Gather in goroutine 37
        /root/go/pkg/mod/github.com/prometheus/client_golang@v1.19.1/prometheus/registry.go:466 +0x568

scross01, 2024-08-15:
> .fetch api key command output
> /moonraker-env/bin/python: No module named lmdb
> have tried to import the module in python3 without success..

Looks like the same issue as #30.  The Moonraker docs are out of date. Try to following to get the API Key

```shell
curl klipperHost:7125/access/api_key
```

scross01, 2024-11-07:
Added improved logging to release v0.12.0

closed 2024-11-07 by scross01