scross01/prometheus-klipper-exporter #37: mcu_task_avg and mcu_task_stddev are not reported

Wulfsta, 2024-10-13:
I see that when mcu metrics [were added](https://github.com/scross01/prometheus-klipper-exporter/commit/595e8d5c196145700d6b29cdacbbdce21bd9e23d) these two were left commented out, and are to this day. Is there a technical reason for this? I would like them to be available given their use to compute and graph the mcu load heuristic.

Wulfsta, 2024-11-06:
I don’t think that applies here; these metrics [are effectively calculated by the mcu](https://github.com/Klipper3d/klipper/blob/eeb2678ec2faa72af57444657b1b4f2589b77d65/src/basecmd.c#L306), and the [variance](https://github.com/Klipper3d/klipper/blob/eeb2678ec2faa72af57444657b1b4f2589b77d65/klippy/mcu.py#L618) from each stats report (at a 1s interval when printing) is [used in a heuristic](https://github.com/Klipper3d/klipper/blob/eeb2678ec2faa72af57444657b1b4f2589b77d65/scripts/graphstats.py#L107) to estimate load. It might be possible to get this data over a longer interval with prometheus, but I’m not sure it would be possible to replicate an accurate load graph like that?

scross01, 2024-11-07:
Fixed in release v0.12.0

Thanks for the PR!

closed 2024-11-07 by scross01