scross01/prometheus-klipper-exporter #41: Add temperature_probe

ANDRIUS-D-ILGUNAS, 2025-03-10:
If only I was better at Golang....

TLDR:  Could you please add `temperature_probe` into the collector?

I see where you collect `temperature_status`, but I have additional temperature probes that I've setup in klipper like this

```
[temperature_probe y_motor]
sensor_type: Generic 3950
sensor_pin: PF5
[temperature_probe e_motor ]
sensor_type: Generic 3950
sensor_pin: PF6
[temperature_probe x_motor]
sensor_type: Generic 3950
sensor_pin: PF7
```

And pointing my browser to `http://my_printer/printer/objects/list`, I see this:
```
{"result": {"objects": ["gcode", "webhooks", "configfile", "mcu", "mcu rpi", "gcode_move", "print_stats", "virtual_sdcard", "pause_resume", "display_status", "stepper_enable", "tmc2209 stepper_x", "tmc2209 stepper_y", "tmc2209 stepper_z", "tmc2209 stepper_z1", "tmc2209 extruder", "heaters", "heater_fan Hot_End", "fan", "controller_fan Right_Case", "fan_generic Left_Case", "exclude_object", "probe", "bltouch", "heater_bed", "bed_mesh", "z_tilt", "screws_tilt_adjust", "filament_switch_sensor switch_sensor", "temperature_probe y_motor", "temperature_probe e_motor ", "temperature_probe x_motor", "motion_report", "query_endstops", "idle_timeout", "system_stats", "manual_probe", "toolhead", "extruder"]}}
```

Further, I if I hit up `/printer/objects/query?temperature_probe%20y_motor`, I get data back:
```
{"result": {"eventtime": 34589.072201753, "status": {"temperature_probe y_motor": {"temperature": 20.574365256655813, "measured_min_temp": 19.93, "measured_max_temp": 20.74, "in_calibration": false, "estimated_expansion": 0, "compensation_enabled": false}}}}
```

So it seems that the exporter _should_ be able to collect this info too, no?  I've tried mimicking `PrinterObjectTemperatureSensor` (mostly by copy-pasting) into `PrinterObjectTemperatureProbe`, but my lack of skill in Golang is blatant.

Would you be willing to add this in for us?

ANDRIUS-D-ILGUNAS, 2025-03-10:
ALSO!  Could we have it (optionally) log to a file please?  If not, then where could the logs be found?  They're not in `/home/pi` or `/var/log`

scross01, 2025-04-05:
I've put initial support for `temperature_probe` in this branch https://github.com/scross01/prometheus-klipper-exporter/tree/temperature_probe 

Can you test and see if this works for you.

scross01, 2025-04-05:
For the logs. If your using the suggested systemd configuration to run the klipper-exporter as a service you can view the logs using

```bash
journalctl -u klipper-exporter
```

The default logging level in the sample `klipper-exporter.service` is set to Warning so you won't get many logs output. To increase the logging level update `/etc/systemd/system/klipper-exporter.service` and change `-logging.level` to Info or Debug, e.g.

```
ExecStart=/home/pi/klipper-exporter/prometheus-klipper-exporter -logging.level Info
```

scross01, 2025-05-18:
Merged to v0.13.0 release

closed 2025-05-18 by scross01